SIU-Robotics/moondawg-ino #37: Updates

BigBroccoli, 2024-06-18:
Combine the libraries into multiple .ino files instead of importable libraries
Restructure the command parser to be more readable

BigBroccoli, 2024-06-18:
https://arduino.stackexchange.com/questions/60656/split-up-arduino-code-into-multiple-files-using-arduino-ide

BigBroccoli, 2024-06-22:
Can also get rid of the lib folder, since Servo is in the Arduino IDE

abarnes6, 2024-09-06:
Misunderstood implementation - reopened

BigBroccoli, 2024-09-09:
I don't think this will work well for our application, most other devs use our current file system

BigBroccoli, 2024-09-10:
> I don't think this will work well for our application, most other devs use our current file system

However let's continue to look into this it could improve dev testing convenience

BigBroccoli, 2024-09-10:
https://arduino.github.io/arduino-cli/1.0/sketch-build-process/#pre-processing

BigBroccoli, 2024-09-17:
> If you use multiple .ino files you need to be aware that the compiler first loads the principal file (the one with the same name as the folder) and then loads the others in alphabetical order.

Variables defined in an earlier file will be visible to code in a file that is loaded later, but not vice-versa.

For my own projects I just put the code into .h files and I never bother with .cpp files.

https://forum.arduino.cc/t/correct-way-to-include-multiple-files/507705/5

BigBroccoli, 2024-09-17:
Ok, from the last comment I decided to "fuck it we ball" and just slam all of the files into the main folder with .ino file
And it seems to have worked

closed 2024-09-21 by BigBroccoli