SIU-Robotics/moondawg-ino #43: Prep code for Arduino alternative

BigBroccoli, 2024-09-18:
Look into how PlatformIO works and what we will need to do to get it working in our use case.

BigBroccoli, 2024-09-21:
Look into ESP32's, ATmega based stuff, etc.
[](https://www.espressif.com/en/products/devkits)
[](https://electronics.stackexchange.com/a/435915)
[](https://platformio.org/)

BigBroccoli, 2024-09-21:
[Guide to Install](https://docs.platformio.org/en/latest/integration/ide/vscode.html#ide-vscode)

BigBroccoli, 2024-09-21:
This tool will be needed to do dev. It has a lot of useful features
[image]
[image]

BigBroccoli, 2024-09-23:
Also need to add platformIO files to repos. Will add once current PR #45 gets merged to main

BigBroccoli, 2024-09-28:
Look at comments from Andrews review of #45 to update comments, Readme, some code, etc.

Marco-Caliendo, 2024-10-01:
Test removing arduino.h from files

Marco-Caliendo, 2024-10-01:
remove externs, put them in main file or something like that

abarnes6, 2024-10-07:
esp wrover kit

abarnes6, 2024-10-07:
[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino

BigBroccoli, 2024-10-10:
Look for better solution to Structs kinda confusing

BigBroccoli, 2024-10-10:
Update Readme based on new changes and on Andrew's comments

BigBroccoli, 2024-10-10:
Make the comments uniform and updated on all code

closed 2024-10-15 by BigBroccoli